Disane87/spoolman-homeassistant #17: "URL does not return JSON content"

Disane87, 2023-10-02:
actually i input the IP (http://192.168.1.133:7912/), and leave API-Key empty and the error i get is "URL does not return JSON content"

_Originally posted by @micro-fx in https://github.com/Disane87/spoolman-homeassistant/discussions/15_

Disane87, 2023-10-02:
When you get the message `URL does not return JSON content` it seems your spoolman really doesn't return the exppected data. You can try to grab your URL and append `api/v1/spool`. Like this (only an example: `https://spoolman.disane.dev/api/v1/spool`)

Now you should get a json with all your spools. In the code I don't do anything else:
https://github.com/Disane87/spoolman-homeassistant/blob/0155d5937b844bb7c4022f5e95a55584ff54a3e2/custom_components/spoolman/coordinator.py#L27

https://github.com/Disane87/spoolman-homeassistant/blob/0155d5937b844bb7c4022f5e95a55584ff54a3e2/custom_components/spoolman/const.py#L9

Hope this helps :)

micro-fx, 2023-10-02:
so i added such as http://192.168.1.133:7912/api/v1/spool
and i get some json format, and includes spool info.

so what is the path i should use when adding the HA integration?

if i use the appended url when adding the integration, i get "**Failed to connect to the URL. Status code: 404**"

Disane87, 2023-10-02:
You only need `http://192.168.1.133:7912` without trailing slash. That should work.

closed 2023-10-02 by Disane87

Disane87, 2023-10-02:
@micro-fx I've implemented a check for missing trailing slashes. Unfortunatly there was a bug. You should install the latest version and it should work.

micro-fx, 2023-10-02:
I can confirm it now works on the new version
[image]

Disane87, 2023-10-02:
Great, thank you for the feedback @micro-fx! Have fun with it! If you have further questions or concerns just hit me up 🆙

commit c4b7b071 (2023-10-02): fix: :bug: Trailing slashes in config are now added when not present / API key removed

fixes #17
closes #19