Disane87/spoolman-homeassistant #44: Can't get intergration to work

Marshalldog, 2023-11-01:
### System Health details

## System Information

version | core-2023.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | *****censored****
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1402
Downloaded Repositories | 42
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | ****Censored**
relayer_connected | true
relayer_region | ap-southeast-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | ap-southeast-1-0.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.1
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 458.4 GB
disk_used | 248.7 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2022.3.1), Mosquitto broker (6.3.1), File editor (5.6.0), Grafana (9.1.1), ModbusTCP2MQTT (0.3.8), Studio Code Server (5.13.0), Home Assistant Google Drive Backup (0.111.1), Check Home Assistant configuration (3.11.0), Hey Ada! (1.2.0), Ring Livestream (1.35), Terminal & SSH (9.7.1), AirCast (3.6.1), Home Panel (2.7.0), Google Assistant SDK (2.5.0), Spotify Connect (0.12.6), Traccar (0.21.1), Samba share (10.0.2), Samba NAS (10.0.2-nas4), Plex Media Server with NAS (2.4.2-dht3), snapcast-server (0.25), Advanced SSH & Web Terminal (15.1.0), CEC Scanner (3.0), librespot (v0.3.1.2), snapcast-client (beta), AppDaemon (0.14.0), AirSonos (3.6.1), Hass.io Community Store (1.30.1), ESPHome (2023.10.5), Home Assistant Google Drive Backup (0.110.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 14
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 21 October 2023 at 12:50 pm
-- | --
current_recorder_run | 1 November 2023 at 4:19 pm
estimated_db_size | 501.68 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Can't get the install to work:
I get to 
[image]
and press submit.
then it just stays as the following, showing no device found
[image]
checked logs and found the following:
```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:359
Integration: Sensor (documentation, issues)
First occurred: 4:19:42 pm (5 occurrences)
Last logged: 4:30:26 pm

Error while setting up spoolman platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 359, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/spoolman/sensor.py", line 46, in async_setup_entry
    spool_device = Spool(hass, coordinator, spool_data, idx, config_entry)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/spoolman/sensor.py", line 75, in __init__
    self._spool["location"] if spool_data["archived"] is False else "Archived"
    ~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'location'

```
Note: I have the correct IP address cause I'm also using a klipper install that feeds data to HASS on the same IP

### Reproduction steps

1.added :https://github.com/Disane87/spoolman-homeassistant to custom repositories under HACS
2.Installed spoolman integration and did the above
3.
...

### Debug logs

```text
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:359
Integration: Sensor (documentation, issues)
First occurred: 4:19:42 pm (5 occurrences)
Last logged: 4:30:26 pm

Error while setting up spoolman platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 359, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/spoolman/sensor.py", line 46, in async_setup_entry
    spool_device = Spool(hass, coordinator, spool_data, idx, config_entry)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/spoolman/sensor.py", line 75, in __init__
    self._spool["location"] if spool_data["archived"] is False else "Archived"
    ~~~~~~~~~~~^^^^^^^^^^^^
KeyError:
```

Disane87, 2023-11-01:
Thank you for this issue @Marshalldog! It looks like the location field is not present in the spool object. What version of spoolman do you use? 

Is it possible that you send me the return of your api for the spools? Just open this url in a browser:

http://your-spoolman/api/v1/spool

the-cyberspace, 2023-11-01:
I had the same problem and after filling up the location the integration in Home Assistant is working now!

Got the latests version from spoolman itself and the integration for HA

Disane87, 2023-11-01:
> I had the same problem and after filling up the location the integration in Home Assistant is working now!
> 
> 
> 
> Got the latests version from spoolman itself and the integration for HA

Thank you for this hint @the-cyberspace! Will consider that the location can be unset in the next version.

If the location is unset I would propose a `unknown` location, so the spools can be grouped into one. 

But yeah, this is a bug which didn't occurred on my end since I have the locations filled 🫣

the-cyberspace, 2023-11-01:
No problem and glad i could help! Didn't had the location setup because i don't use that option and now there all just "home" ...

But i like the idea to group them!

Disane87, 2023-11-01:
:tada: This issue has been resolved in version 0.2.1-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.2.1-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2023-11-01:
@the-cyberspace @Marshalldog I've created a dev release for this issue where this problem should be fixed. Would be nice if you can test this on your end.

Marshalldog, 2023-11-02:
it worked! thx

Disane87, 2023-11-02:
If it's working now, it's totally enough for me.  Will consider this as resolved by now. If you find more, just open another issue. 

Thank you for contributing!

Disane87, 2023-11-02:
:tada: This issue has been resolved in version 0.2.2-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.2.2-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2023-11-02:
:tada: This issue has been resolved in version 0.2.2 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.2.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 8ce5abd6 (2023-11-01): fix: :bug: fixed a bug where location from spoolman is empty

fixes #44
commit 33949b6b (2023-11-02): fix: interims commit

It doesn't really fix #44 but that commit should trigger a new version, because the build system was broken unfortunatly.

fixes #44