Disane87/spoolman-homeassistant #60: Not find

Eloura74, 2023-11-20:
### System Health details

[image]
I can't get the spoolman link to work

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

<

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
<
```

### Diagnostics dump

  <

Disane87, 2023-11-20:
You need to use this repo to add in hacs, not the spool endpoint of Spoolman. Please check out the docs to HACS how to install custom integrations.

Eloura74, 2023-11-20:
ok thank

Disane87, 2023-11-20:
I've done this issue since it's not a problem of this integration. If you have further question, please open a discussion or (if it's a bug) an issue. Thank!

closed 2023-11-20 by Disane87