Disane87/spoolman-homeassistant #62: Do you have an example of a nice map proposed for the dashboard?

Disane87, 2023-11-27:
Do you have an example of a nice map proposed for the dashboard?

_Originally posted by @Eloura74 in https://github.com/Disane87/spoolman-homeassistant/issues/61#issuecomment-1827637967_

Disane87, 2023-12-08:
:tada: This issue has been resolved in version 0.2.2-dev.2 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.2.2-dev.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

pdsccode, 2024-01-04:
Just for info as I struggled a bit implementing the suggested card:

I like to sort them properly by remaining weight so I added a 'numeric: true' to the sorting options:

`sort:
  method: state
  reverse: true`

commit c99262f4 (2023-12-08): docs: added usage of entities in cards to readme

closes #62
closed 2024-02-01 by Disane87