Disane87/spoolman-homeassistant #68: Error doing job: Task exception was never retrieved

MacVille, 2023-12-13:
### System Health details

## System Information

version | core-2023.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.63-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4993
Installed Version | 1.33.0
Stage | running
Available Repositories | 1422
Downloaded Repositories | 63

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.2
-- | --
update_channel | stable
supervisor_version | supervisor-2023.11.6
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 62.3 GB
disk_used | 23.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 55
views | 40
mode | storage

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Error on Logfile with the Following Error:

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: homeassistant
Source: custom_components/spoolman/sensor.py:130
Integration: Spoolman (documentation, issues)
First occurred: 00:28:23 (4 occurrences)
Last logged: 01:13:23

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 243, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 399, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 96, in _handle_coordinator_update
    self.check_for_threshold(self._spool, self._spool["used_percentage"])
  File "/config/custom_components/spoolman/sensor.py", line 130, in check_for_threshold
    self._spool["location"],
    ~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'location'

### Reproduction steps

1. Install integration
2. Add integration (w/ specifying the spoolman host)
3. Integration is running fine

### Debug logs

```text
No Debug Log avalibale
```

Disane87, 2023-12-13:
Could you please use the latest dev release? There are some bug fixes for fields that are not handled correctly when they don't exists. Do you already checked to service location in spoolman and see if that fix this error?

MacVille, 2023-12-13:
I'm using the Version 0.2.2.
I doesn't have defined any Location because that Field is not Mandatory on Spoolman.
I will give it a shot and if this fixes the Issue will come back.

Disane87, 2023-12-13:
Thank you :) the dev version should be more reliable than the current stable. If you have success using that version I would release the dev version properly.

MacVille, 2023-12-21:
Sorry for the late Responce. I have tested it with the Release Version and after adding the Location the Error Message is gone.
But would it not be better, that this Field is not Mandatory for the Configuration? Because on Spoolman itself this is an Non-Mandatory Field.

Disane87, 2023-12-21:
Because of that the location is in the latest dev release not mandatory anymore. 

Did you test the latest dev release?

the-cyberspace, 2024-02-01:
Don't want to hijack something, but upgraded to the latest dev and i'm still get these errors:

```
Logger: homeassistant
Source: custom_components/spoolman/sensor.py:111
Integration: Spoolman (documentation, issues)
First occurred: 13:36:06 (4 occurrences)
Last logged: 14:21:06

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 256, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 412, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 183, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 111, in _handle_coordinator_update
    if self._filament["used_weight"] is None:
       ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'used_weight
```

Disane87, 2024-02-01:
Will open a new issue for that particular case :)

the-cyberspace, 2024-02-01:
Ahh you got me there. I didn"t see the last part and i`ve got used_weight instead of location. Stupid me. Thanks for creating a new issue!

Disane87, 2024-03-05:
probably fixed with #77 please test

closed 2024-03-05 by Disane87