Disane87/spoolman-homeassistant #71: Don't use that many State Attributes

mawoka-myblock, 2023-12-26:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The usage of state attributes could make the database size increase quickly.

### Describe the solution you'd like

Create dedicated sensors, especially for remaining and used weight, since that is update pretty frequently

### Describe alternatives you've considered

Maybe not that important?

### Additional context

https://developers.home-assistant.io/docs/core/entity/#generic-properties

Disane87, 2023-12-26:
That's a good hint, will dig into this. Thank you!

mawoka-myblock, 2024-03-08:
maybe create a "device" per spool?

mawoka-myblock, 2024-03-08:
Oh, I'd also suggest removing the good first issue label, as I'd say that this isn't one.

Disane87, 2024-03-08:
> maybe create a "device" per spool?

That could be indeed a reasonable alternative

Disane87, 2024-05-21:
Currently I'm evaluating the possibility to create "Hubs" für every storage location and within that a device per spool and for every attribute an own entity. Something like Meross Integration is doing here:
[image]

But at the moment we're pretty busy with renovating our kitchen so that need some time. But with that refactoring I gonna address this problem #103 as well.

mawoka-myblock, 2024-05-21:
tbh, I don't even use your integration, just stumbled upon it, so take your time. If you should still need help, feel free to ask

Disane87, 2025-11-27:
:tada: This issue has been resolved in version 1.2.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v1.2.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket: