Disane87/spoolman-homeassistant #77: self._filament["used_weight"] is None:

Disane87, 2024-02-01:
Don't want to hijack something, but upgraded to the latest dev and i'm still get these errors:

```
Logger: homeassistant
Source: custom_components/spoolman/sensor.py:111
Integration: Spoolman (documentation, issues)
First occurred: 13:36:06 (4 occurrences)
Last logged: 14:21:06

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 256, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 412, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 183, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 111, in _handle_coordinator_update
    if self._filament["used_weight"] is None:
       ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'used_weight
```

edit: Tested on HA 2024.1.6 and 2024.2.0b0

_Originally posted by @the-cyberspace in https://github.com/Disane87/spoolman-homeassistant/issues/68#issuecomment-1921324037_

Disane87, 2024-02-01:
@the-cyberspace
It looks like your `used_weight` is empty. Could you doublecheck that?

If you create a new spool in spoolman, there should be a 0 for that by default:
[image]

Disane87, 2024-02-01:
OK thanks. Will check that! 

Thank you for reporting 🥇

Disane87, 2024-02-01:
Ah I see the exception seems to be cut off:

> KeyError: 'used_weight

Can you please post the complete exception?

the-cyberspace, 2024-02-01:
Sure, but it's a looooong list and created a .json filet:

[spool.json](https://github.com/Disane87/spoolman-homeassistant/files/14127105/spool.json)

the-cyberspace, 2024-02-01:
You mean this?

```
2024-02-01 13:34:10.841 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 243, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 399, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 111, in _handle_coordinator_update
    if self._filament["used_weight"] is None:
       ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'used_weight'
2024-02-01 13:49:10.840 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 243, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 399, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 111, in _handle_coordinator_update
    if self._filament["used_weight"] is None:
       ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'used_weight'
2024-02-01 14:00:00.146 WARNING (MainThread) [homeassistant.helpers.script.trigger_update_coordinator] Trigger Update Coordinator: Running script requires passing in a context
2024-02-01 14:00:00.188 WARNING (MainThread) [homeassistant.helpers.script.trigger_update_coordinator] Trigger Update Coordinator: Running script requires passing in a context
2024-02-01 14:00:00.190 ERROR (MainThread) [homeassistant.helpers.sensor] Error rendering availability template for sensor.temperature_forecast_next_hour: UndefinedError: 'forecast' is undefined
2024-02-01 14:04:10.841 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 243, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 399, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 111, in _handle_coordinator_update
    if self._filament["used_weight"] is None:
       ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'used_weight'
2024-02-01 14:19:10.856 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 243, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 399, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 111, in _handle_coordinator_update
    if self._filament["used_weight"] is None:
       ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'used_weight'
2024-02-01 14:34:10.841 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 243, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 399, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 111, in _handle_coordinator_update
    if self._filament["used_weight"] is None:
       ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'used_weight'
2024-02-01 14:37:47.011 WARNING (MainThread) [py.warnings] /usr/local/lib/python3.11/site-packages/turbojpeg.py:901: UserWarning: Corrupt JPEG data: 1 extraneous bytes before marker 0xd6
  warnings.warn(self.__get_error_string(handle))

2024-02-01 14:38:36.552 WARNING (MainThread) [py.warnings] /usr/local/lib/python3.11/site-packages/turbojpeg.py:901: UserWarning: Corrupt JPEG data: 1 extraneous bytes before marker 0xd4
[... 75 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 399, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 111, in _handle_coordinator_update
    if self._filament["used_weight"] is None:
       ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'used_weight'
[... cut ...]
2024-02-01 16:10:10.984 DEBUG (MainThread) [custom_components.spoolman.coordinator] Finished fetching Spoolman data data in 0.006 seconds (success: True)
2024-02-01 16:10:10.985 DEBUG (MainThread) [custom_components.spoolman.coordinator] SpoolManCoordinator.__init__
2024-02-01 16:10:10.985 DEBUG (MainThread) [custom_components.spoolman.coordinator] SpoolManCoordinator._async_update_data
2024-02-01 16:10:10.991 DEBUG (MainThread) [custom_components.spoolman.coordinator] Finished fetching Spoolman data data in 0.006 seconds (success: True)
```

Disane87, 2024-02-01:
Yeah thanks!

Disane87, 2024-02-06:
@the-cyberspace thank you for your patience! I've tested this and tried to reproduce this issue. I really can't reproduce it. All my spools have `used_weight` in the result of the api and is available in runtime.

I could make it more robust and don't exit unhandled if properties are not set. But that's just a band-aid on the wound that would hide deeper problems. IMHO there shouldn't be any problems here.

Is it possible that you get me access to your spoolman db or export it to me so I can setup a new instance with put database to investigate it?

Disane87, 2024-02-06:
I've added these changes. Additionally I've integrated some loggin into the API class. It seems the api sometimes returns with these missing properties but I currently can't say whats the problem here.

If you encounter problems while calculating the property `used_percentage` then have a look into the debug log. There should be now more information  whats going on.

Disane87, 2024-02-06:
:tada: This issue has been resolved in version 0.3.0-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.3.0-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

the-cyberspace, 2024-02-06:
Thanks for the update! Now i'm seeing these errors:
```
Logger: custom_components.spoolman.sensor
Source: custom_components/spoolman/sensor.py:115
Integration: Spoolman (documentation, issues)
First occurred: 22:40:55 (24 occurrences)
Last logged: 22:55:55

SpoolManCoordinator: Spool with ID '4' has no 'used_weight' set or property is missing in filament. Can't calculate usage. Skipping.
SpoolManCoordinator: Spool with ID '1' has no 'used_weight' set or property is missing in filament. Can't calculate usage. Skipping.
SpoolManCoordinator: Spool with ID '14' has no 'used_weight' set or property is missing in filament. Can't calculate usage. Skipping.
SpoolManCoordinator: Spool with ID '9' has no 'used_weight' set or property is missing in filament. Can't calculate usage. Skipping.
SpoolManCoordinator: Spool with ID '7' has no 'used_weight' set or property is missing in filament. Can't calculate usage. Skipping.
```

```
2024-02-06 23:07:52.569 DEBUG (MainThread) [custom_components.spoolman] __init__.async_unload_entry
2024-02-06 23:07:52.573 DEBUG (MainThread) [custom_components.spoolman] __init__.async_setup_entry
2024-02-06 23:07:52.573 DEBUG (MainThread) [custom_components.spoolman.coordinator] SpoolManCoordinator.__init__
2024-02-06 23:07:52.573 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: __init__
2024-02-06 23:07:52.573 DEBUG (MainThread) [custom_components.spoolman.coordinator] SpoolManCoordinator._async_update_data
2024-02-06 23:07:52.573 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: get_spool
2024-02-06 23:07:52.573 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: string_from_dictionary
2024-02-06 23:07:52.580 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: get_spool response [{'id': 11, 'registered': '2023-10-06T09:24:00.198178Z', 'first_used': '2023-10-14T05:43:51Z', 'last_used': '2024-02-04T13:33:16Z', 'filament': {'id': 12, 'registered': '2023-10-06T09:23:41.074348Z', 'name': 'Army green', 'vendor': {'id': 1, 'registered': '2023-08-13T21:40:28.789541Z', 'name': 'Polymaker', 'extra': {}}, 'material': 'ASA', 'price': 37.95, 'density': 1.13, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 147.0, 'article_number': 'SKU: PF01030', 'comment': 'Acrylonitrile Styrene Acrylate', 'settings_extruder_temp': 250, 'settings_bed_temp': 100, 'color_hex': '085C08', 'extra': {}}, 'remaining_weight': 917.7415395785947, 'used_weight': 82.25846042140529, 'remaining_length': 337657.16309556726, 'used_length': 30264.684759997253, 'location': '', 'archived': False, 'extra': {}}, {'id': 12, 'registered': '2023-10-09T08:40:06Z', 'first_used': '2023-11-16T13:17:37Z', 'last_used': '2023-11-16T16:04:38Z', 'filament': {'id': 13, 'registered': '2023-10-09T08:39:28Z', 'name': 'Green - 500g', 'vendor': {'id': 2, 'registered': '2023-08-13T22:01:37.278411Z', 'name': 'Real Filament', 'extra': {}}, 'material': 'PLA', 'price': 34.5, 'density': 1.24, 'diameter': 1.75, 'weight': 500.0, 'spool_weight': 169.0, 'settings_extruder_temp': 210, 'settings_bed_temp': 50, 'color_hex': '07A90E', 'extra': {}}, 'remaining_weight': 32.0, 'used_weight': 468.0, 'remaining_length': 10729.07582133646, 'used_length': 156912.73388704576, 'location': '', 'archived': False, 'extra': {}}, {'id': 10, 'registered': '2023-09-08T15:56:17.472912Z', 'first_used': '2023-09-10T12:19:07.407003Z', 'last_used': '2023-10-31T20:40:53Z', 'filament': {'id': 11, 'registered': '2023-09-08T15:55:47.144274Z', 'name': 'Brown', 'vendor': {'id': 2, 'registered': '2023-08-13T22:01:37.278411Z', 'name': 'Real Filament', 'extra': {}}, 'material': 'ABS', 'price': 29.5, 'density': 1.05, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 147.0, 'comment': 'Acrylonitrile Butadiene Styrene', 'settings_extruder_temp': 240, 'settings_bed_temp': 100, 'color_hex': '693105', 'extra': {}}, 'remaining_weight': 754.0, 'used_weight': 246.0, 'remaining_length': 298549.3074379981, 'used_length': 97404.68120656173, 'location': '', 'archived': False, 'extra': {}}, {'id': 2, 'registered': '2023-08-13T22:15:32.089018Z', 'first_used': '2023-08-13T10:15:32Z', 'last_used': '2023-09-26T13:31:13.905703Z', 'filament': {'id': 3, 'registered': '2023-08-13T22:14:15.504888Z', 'name': 'Green', 'vendor': {'id': 1, 'registered': '2023-08-13T21:40:28.789541Z', 'name': 'Polymaker', 'extra': {}}, 'material': 'ASA', 'price': 37.95, 'density': 1.13, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 147.0, 'article_number': 'SKU: PF01030', 'comment': 'Acrylonitrile Styrene Acrylate', 'settings_extruder_temp': 250, 'settings_bed_temp': 100, 'color_hex': '00B300', 'extra': {}}, 'remaining_weight': 70.0, 'used_weight': 930.0, 'remaining_length': 25754.529349889515, 'used_length': 342167.318505675, 'location': '', 'archived': False, 'extra': {}}, {'id': 13, 'registered': '2023-10-09T08:41:42Z', 'filament': {'id': 14, 'registered': '2023-10-09T08:41:19Z', 'name': 'Khaki grey - 500g', 'vendor': {'id': 2, 'registered': '2023-08-13T22:01:37.278411Z', 'name': 'Real Filament', 'extra': {}}, 'material': 'PLA', 'price': 34.5, 'density': 1.24, 'diameter': 1.75, 'weight': 500.0, 'spool_weight': 169.0, 'settings_extruder_temp': 210, 'settings_bed_temp': 50, 'color_hex': '514F4F', 'extra': {}}, 'remaining_weight': 106.0, 'used_weight': 394.0, 'remaining_length': 35540.06365817703, 'used_length': 132101.74605020517, 'location': '', 'archived': False, 'extra': {}}, {'id': 5, 'registe [... cut ...] ter': 1.75, 'weight': 1000.0, 'spool_weight': 219.0, 'settings_extruder_temp': 210, 'settings_bed_temp': 50, 'color_hex': '898686', 'extra': {}}, 'remaining_weight': 38.572614793271555, 'used_weight': 961.4273852067284, 'remaining_length': 12932.765898256715, 'used_length': 322350.8535185077, 'location': '', 'archived': False, 'extra': {}}]
2024-02-06 23:07:52.587 DEBUG (MainThread) [custom_components.spoolman.coordinator] Finished fetching Spoolman data data in 0.006 seconds (success: True)
```

Disane87, 2024-02-06:
Thank you! Will dig into this. Hopefully I see some hints where this issue comes from. 

At the moment I totally have no clue. We'll see. 

Hit you up when I need further information.

Disane87, 2024-02-07:
Are there more information? I'm missing these lines in the logs:
https://github.com/Disane87/spoolman-homeassistant/blob/ffb598348804316f7f50090586b44c8e9caa6ad8/custom_components/spoolman/sensor.py#L107-L108

At the moment I guess these properties are missing sometimes when the coordinator reloads. I've had this just now but after it occurs, it never happend again. This is a pretty strange bug. I suppose I will need more time and information.

Hope you didn't mind if I'm getting back to you with requests.

the-cyberspace, 2024-02-07:
You mean this?

```
2024-02-07 10:30:31.957 DEBUG (MainThread) [custom_components.spoolman] __init__.async_unload_entry
2024-02-07 10:30:31.958 DEBUG (MainThread) [custom_components.spoolman] __init__.async_setup_entry
2024-02-07 10:30:31.958 DEBUG (MainThread) [custom_components.spoolman.coordinator] SpoolManCoordinator.__init__
2024-02-07 10:30:31.959 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: __init__
2024-02-07 10:30:31.959 DEBUG (MainThread) [custom_components.spoolman.coordinator] SpoolManCoordinator._async_update_data
2024-02-07 10:30:31.959 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: get_spool
2024-02-07 10:30:31.959 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: string_from_dictionary
2024-02-07 10:30:31.966 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: get_spool response [{'id': 11, 'registered': '2023-10-06T09:24:00.198178Z', 'first_used': '2023-10-14T05:43:51Z', 'last_used': '2024-02-07T08:30:06Z', 'filament': {'id': 12, 'registered': '2023-10-06T09:23:41.074348Z', 'name': 'Army green', 'vendor': {'id': 1, 'registered': '2023-08-13T21:40:28.789541Z', 'name': 'Polymaker', 'extra': {}}, 'material': 'ASA', 'price': 37.95, 'density': 1.13, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 147.0, 'article_number': 'SKU: PF01030', 'comment': 'Acrylonitrile Styrene Acrylate', 'settings_extruder_temp': 250, 'settings_bed_temp': 100, 'color_hex': '085C08', 'extra': {}}, 'remaining_weight': 917.3338443007136, 'used_weight': 82.66615569928638, 'remaining_length': 337507.16309556726, 'used_length': 30414.684759997253, 'location': '', 'archived': False, 'extra': {}}, {'id': 12, 'registered': '2023-10-09T08:40:06Z', 'first_used': '2023-11-16T13:17:37Z', 'last_used': '2023-11-16T16:04:38Z', 'filament': {'id': 13, 'registered': '2023-10-09T08:39:28Z', 'name': 'Green - 500g', 'vendor': {'id': 2, 'registered': '2023-08-13T22:01:37.278411Z', 'name': 'Real Filament', 'extra': {}}, 'material': 'PLA', 'price': 34.5, 'density': 1.24, 'diameter': 1.75, 'weight': 500.0, 'spool_weight': 169.0, 'settings_extruder_temp': 210, 'settings_bed_temp': 50, 'color_hex': '07A90E', 'extra': {}}, 'remaining_weight': 32.0, 'used_weight': 468.0, 'remaining_length': 10729.07582133646, 'used_length': 156912.73388704576, 'location': '', 'archived': False, 'extra': {}}, {'id': 10, 'registered': '2023-09-08T15:56:17.472912Z', 'first_used': '2023-09-10T12:19:07.407003Z', 'last_used': '2023-10-31T20:40:53Z', 'filament': {'id': 11, 'registered': '2023-09-08T15:55:47.144274Z', 'name': 'Brown', 'vendor': {'id': 2, 'registered': '2023-08-13T22:01:37.278411Z', 'name': 'Real Filament', 'extra': {}}, 'material': 'ABS', 'price': 29.5, 'density': 1.05, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 147.0, 'comment': 'Acrylonitrile Butadiene Styrene', 'settings_extruder_temp': 240, 'settings_bed_temp': 100, 'color_hex': '693105', 'extra': {}}, 'remaining_weight': 754.0, 'used_weight': 246.0, 'remaining_length': 298549.3074379981, 'used_length': 97404.68120656173, 'location': '', 'archived': False, 'extra': {}}, {'id': 2, 'registered': '2023-08-13T22:15:32.089018Z', 'first_used': '2023-08-13T10:15:32Z', 'last_used': '2023-09-26T13:31:13.905703Z', 'filament': {'id': 3, 'registered': '2023-08-13T22:14:15.504888Z', 'name': 'Green', 'vendor': {'id': 1, 'registered': '2023-08-13T21:40:28.789541Z', 'name': 'Polymaker', 'extra': {}}, 'material': 'ASA', 'price': 37.95, 'density': 1.13, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 147.0, 'article_number': 'SKU: PF01030', 'comment': 'Acrylonitrile Styrene Acrylate', 'settings_extruder_temp': 250, 'settings_bed_temp': 100, 'color_hex': '00B300', 'extra': {}}, 'remaining_weight': 70.0, 'used_weight': 930.0, 'remaining_length': 25754.529349889515, 'used_length': 342167.318505675, 'location': '', 'archived': False, 'extra': {}}, {'id': 13, 'registered': '2023-10-09T08:41:42Z', 'filament': {'id': 14, 'registered': '2023-10-09T08:41:19Z', 'name': 'Khaki grey - 500g', 'vendor': {'id': 2, 'registered': '2023-08-13T22:01:37.278411Z', 'name': 'Real Filament', 'extra': {}}, 'material': 'PLA', 'price': 34.5, 'density': 1.24, 'diameter': 1.75, 'weight': 500.0, 'spool_weight': 169.0, 'settings_extruder_temp': 210, 'settings_bed_temp': 50, 'color_hex': '514F4F', 'extra': {}}, 'remaining_weight': 106.0, 'used_weight': 394.0, 'remaining_length': 35540.06365817703, 'used_length': 132101.74605020517, 'location': '', 'archived': False, 'extra': {}}, {'id': 5, 'registered': '2023-08-22T17:46:31.374085Z', 'first_used': '2023-09-11T10:52:57.320019Z', 'last_used': '2023-09-11T13:18:27.400888Z', 'filament': {'id': 5, 'registered': '2023-08-22T17:44:50.492129Z', 'name': 'Black', 'vendor': {'id': 2, 'registered': '2023-08-13T22:01:37.278411Z', 'name': 'Real Filament', 'extra': {}}, 'material': 'PLA', 'price': 34.5, 'density': 1.24, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 219.0, 'settings_extruder_temp': 210, 'settings_bed_temp': 50, 'color_hex': '000000', 'extra': {}}, 'remaining_weight': 22.0, 'used_weight': 978.0, 'remaining_length': 7376.239627168818, 'used_length': 327907.3797895956, 'location': '', 'archived': False, 'extra': {}}, {'id': 6, 'registered': '2023-08-22T17:47:15.498437Z', 'first_used': '2023-09-28T12:07:05.637207Z', 'last_used': '2023-12-16T14:22:57Z', 'filament': {'id': 6, 'registered': '2023-08-22T17:46:58.705410Z', 'name': 'White', 'vend [... cut ...] l Filament', 'extra': {}}, 'material': 'PLA', 'price': 34.5, 'density': 1.24, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 219.0, 'settings_extruder_temp': 210, 'settings_bed_temp': 50, 'color_hex': '898686', 'extra': {}}
2024-02-07 11:31:07.491 WARNING (MainThread) [custom_components.spoolman.sensor] SpoolManCoordinator: Spool with ID '7' has no 'used_weight' set or property is missing in filament. Can't calculate usage. Skipping.
```

Just had "debug logging" enabled in the integration.

Disane87, 2024-03-03:
For your info:
I'm on it but pretty busy with daily life. If I have something new I gonna hit you up.

Disane87, 2024-03-04:
:tada: This issue has been resolved in version 0.3.0-dev.3 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.3.0-dev.3)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2024-03-04:
@the-cyberspace I've found the bug but I need some input of you if it works on your end. If it works, I gonna push it to main. Thank you!

the-cyberspace, 2024-03-05:
Just upgraded to 0.3.0-dev.3 and there are no errors in the logs anymore as far as i can see. Thanks for the hard work!

Disane87, 2024-03-05:
Great thank you! Just have a watch if the `used_weight` is decremented when you are printing :)

Disane87, 2024-03-05:
:tada: This issue has been resolved in version 0.3.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.3.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit df845215 (2024-02-06): feat: :bug: Add debug logging for spool information

Add debug log messages to provide insight into the spool and filament
information in the `Spool` class. This helps in troubleshooting and
understanding the data being processed. The added logs improve the
clarity of the code and facilitate debugging in case of any issues.

Fixes: #77
commit 03ce8fc1 (2024-03-04): fix: :bug: Fixed a bug where `used_weight` is empty sometimes

fixes #77
closed 2024-03-05 by Disane87