Disane87/spoolman-homeassistant #80: Repository structure for v0.16.1 is not compliant

sweidinger, 2024-02-25:
### System Health details

## System Information

version | core-2024.2.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.1.74-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4911
Installed Version | 1.34.0
Stage | running
Available Repositories | 1399
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.5
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 30.8 GB
disk_used | 5.1 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (17.1.1), File editor (5.8.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 20. Februar 2024 um 06:51
-- | --
current_recorder_run | 25. Februar 2024 um 20:50
estimated_db_size | 456.52 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

If I want to add this repo to the custom repo's in HACS, I got an error "Repository structure for v0.16.1 is not compliant".

### Reproduction steps

1.HACS
2.Integrations
3.Custom Repositories
...

### Debug logs

```text
No Log because can't install.
```

Disane87, 2024-02-25:
Thank you for reporting! Will check that out! Gonna hit you up if I need something

Disane87, 2024-02-27:
@sweidinger What HACS version do you use? My HACS version is `1.34.0`
[image]

I'm pretty curious what this `v0.16.1` is.

Disane87, 2024-03-05:
I've tested this with a fresh HA installation and newst HACS. Thats working fine.

Disane87, 2024-03-08:
@sweidinger I have an idead what do you. Is it possible you have added the spoolman repo to HACS and not this? The version of the error message `Repository structure for v0.16.1 is not compliant` indicates that because Spoolman's current version is `v0.16.1`. Please try to add my repo to HACS and try installing it again. 

I will close this by now.

closed 2024-03-08 by Disane87

sweidinger, 2024-03-08:
I have no idea, why I got this error message, tried it again, to add the repo and it worked. Its magic.... ;-)