Disane87/spoolman-homeassistant #100: Can't setup integration without clipper

elemakil, 2024-03-17:
### System Health details

## System Information

version | core-2024.3.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1415
Downloaded Repositories | 95
HACS Data | ok

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 16

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 156.8 GB
disk_used | 126.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (17.2.0), File editor (5.8.0), ESPHome (2024.2.2), Studio Code Server (5.15.0), Log Viewer (0.17.0), TasmoAdmin (0.29.1), Cloudflared (5.1.6), JupyterLab (0.14.3), Zigbee2MQTT (1.36.0-1), Crowdsec (1.6.0), Crowdsec Firewall Bouncer (v0.0.28), Music Assistant BETA (2.0.0b111)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 65
views | 18
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 17 February 2024 at 11:55
-- | --
current_recorder_run | 17 March 2024 at 17:40
estimated_db_size | 8521.76 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.17
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 1
total_devices | 1

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I have just installed spoolman-homeassistant v0.4.1 using HACS and can't actually perform the initial configuration of the integration. I have a working spoolman instance without having klipper.

When I browse to the integrations page and try to add it using the "Add Integration" button and enter the spoolman instance URL, I get the error message "Error testing klipper url: /server/info":

[image]

Removing the trailing slash, port or protocol for the spoolman host address does not change the error message (and would likely result in connection failure for two out of those three anyways...).

I can confirm that the spoolman instance works and is reachable from the HomeAssistant host as executing

````
curl http://192.168.179.11:7912/api/v1/spool
````

from within a terminal on the HostAssistant host produces valid JSON output listing the registered spools.

I had a quick look at the source code and I assume that [this](https://github.com/Disane87/spoolman-homeassistant/blob/20ba583317945841822842887c9feffa08735a0f/custom_components/spoolman/config_flow.py#L46) bit of code is not working as intended and always attempts to communicate with klipper even when a klipper URL is not configured (the "is not None" check should probably check for truthyness instead).

### Reproduction steps

1. Install spoolman-homeassistant (v0.4.1)
2. Try to configure the integration without having a clipper instance

### Debug logs

```text
likely not relevant / debug logs are not available for integration since it is not configured yet
```

Disane87, 2024-03-18:
Thanks for the find! Found and fixed the issue :) Whgen the build is done, you can check that ageinst a dev build. Would be nice if you give me some feedback if it works on your end, so I can push it to main.

Disane87, 2024-03-18:
:tada: This issue has been resolved in version 0.4.2-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.4.2-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

elemakil, 2024-03-18:
I can confirm that I was successful in adding the integration in v0.4.2-dev.1

Thanks for the swift resolution!

Disane87, 2024-03-18:
Thank you for the fast reply! Will push it to main later 🙂 

Thank you for reporting this

Disane87, 2024-03-18:
:tada: This issue has been resolved in version 0.4.2 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.4.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 441e4e58 (2024-03-18): fix: :bug: Fixes a bug where klipper url wasn't checked correctly which leads to an exception

fixes #100
closed 2024-03-18 by Disane87