Disane87/spoolman-homeassistant #103: Spools get shuffled around when a spool is archived

hvraven, 2024-03-21:
### System Health details

## System Information

version | core-2024.3.0
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.77
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 2
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 20. März 2024 um 08:22
-- | --
current_recorder_run | 21. März 2024 um 08:19
estimated_db_size | 3979.16 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I observe that the integration gets confused when spools are archived in spoolman while Home Assistant is running. It appears that the IDs no longer match the spool. The picture below shows one of my spools in storage. Each jump in the remaining amount is when a spool was archived. The displayed color also no longer matches the spool. When I restart home assistant the assignment is corrected, instead the archived spools are marked as no longer existent 

[image]

### Reproduction steps

1. Add some spools
2. Archive a spool

### Debug logs

```text
follow later when I have a new spool to archive.
```

Disane87, 2024-03-22:
Thank you for filing that issue and your investigation! I guess this issue seems to be related with reloading the spools (entities) in HA when something has changed. 

I need to reproduce this and hope I can encounter this problem. 

Actually I'm trying to handle a complete reload the spools from Spoolman automatically because these oddities are mostly gone if HA is restarted or the integration is reloaded manually (in the integration settings). Would be nice if you can test this with your issue if it's gone after that. If that works and the problem is gone, I have to find a way to reload the spools periodically new with all current data.

b2un0, 2025-02-27:
the main problem here is, that the HA `entity_id` does not contain the unique spool id from spoolman.
Currently, its only the spool name.

i also had some issues if i archive a spool and replace them with a same model same name).

Disane87, 2025-03-09:
:tada: This issue has been resolved in version 1.0.0-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v1.0.0-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2025-03-09:
:tada: This issue has been resolved in version 1.0.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v1.0.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2025-03-09:
Should be fixed with #194. Please check that with the latest version. You can reopen if its not working for you :) Tahnk you!

PR #194 (2025-03-08): Automatic updates for names and locations
I've been working on making automations for my Bambu Lab P1S, and have found a few issues that I would like to fix.

- Updates names of entities without requiring reload
- Updates devices of entities without requiring reload 
- The unique_id of the sensors use a string that won't change when the filament name changes (was creating new entities when the name/material changed)
- The initial entity_id uses the ID instead of name.
- Do not require name/material for sensor, defaulting to the new entity_id format

Unique_id change Might fix #103 and #174

Th [... cut]
closed 2025-03-09 by Disane87