Disane87/spoolman-homeassistant #105: Cant configure integration without active spool

HermanCattaneo, 2024-03-29:
### System Health details

## System Information

version | core-2024.3.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20-haos
arch | x86_64
timezone | Europe/Budapest
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1401
Downloaded Repositories | 33

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 9 April 2024 at 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-2.ui.nabu.casa
certificate_status | ready
instance_id | 3b98053420a34df7bdafe861765b3da3
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 30.8 GB
disk_used | 13.6 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | TasmoAdmin (0.30.0), Duck DNS (1.16.0), File editor (5.8.0), Mosquitto broker (6.4.0), Terminal & SSH (9.10.0), ESPHome (2021.11.3), Double Take Proxy (1.0.0), NGINX Home Assistant SSL proxy (3.9.0), Swatch (3.1.0-2717ec6)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 19
views | 19
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 14 March 2024 at 06:40
-- | --
current_recorder_run | 29 March 2024 at 23:56
estimated_db_size | 1629.82 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Sonoff</summary>

version | 3.6.0 (8dd8af9)
-- | --
cloud_online | 3 / 7
local_online | 3 / 3

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

If there is no active spool selected, cant configure the integration, because active_spool_id cant be empty.

### Reproduction steps

1. Fresh spoolman install
2. Create vendor, filament, spool(s)
3. Dont select any spool (in fluidd)
4. Add intregration and try configure it
...

### Debug logs

```text
2024-03-30 00:13:47.025 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up spoolman platform for sensor
Traceback (most recent call last):
  File "/config/custom_components/spoolman/coordinator.py", line 61, in _async_update_data
    klipper_active_spool: int = await KlipperAPI(klipper_url).get_active_spool_id()
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/spoolman/classes/klipper_api.py", line 23, in get_active_spool_id
    return int(data.get('result').get("spool_id"))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
```

Disane87, 2024-03-30:
I've fixed this issue. Could you please test it with the latest dev version? If it's fixed, I would poush it to main :) 
Thank you in advance!

HermanCattaneo, 2024-04-01:
@Disane87 i selected in hacs the v0.4.2-dev.2 version, inactivated the active spool

```
20:51:34.130 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up spoolman platform for sensor
Traceback (most recent call last):
  File "/config/custom_components/spoolman/coordinator.py", line 61, in _async_update_data
    klipper_active_spool: int | None = await KlipperAPI(klipper_url).get_active_spool_id()
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/spoolman/classes/klipper_api.py", line 23, in get_active_spool_id
    # Extract the spool_id from the data dictionary
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 318, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/spoolman/coordinator.py", line 71, in _async_update_data
    except Exception as exception:
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant.helpers.update_coordinator.UpdateFailed: Error fetching data from API: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 350, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/spoolman/sensor.py", line 41, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in async_config_entry_first_refresh
    raise ex
homeassistant.exceptions.ConfigEntryNotReady: Error fetching data from API: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

```

Disane87, 2024-04-01:
Thank you! Nah the workflow will close it when I push it to main

closed 2024-04-01 by Disane87