Disane87/spoolman-homeassistant #119: Update the "remaining weight" (and other fields) from HA

bavala3010, 2024-06-01:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

HA cannot update/overwrite the value of attributes. The entities are presented as "read-only". 

### Describe the solution you'd like

I would like to update some values in the Spoolman database via HA, by overwriting/updating some attributes.
Would that be possible via a "service" specifically created for spoolman?

### Describe alternatives you've considered

Updating the Spoolman database from HA

### Additional context

Since HA knows which spool is currently be used and it knows which how many grams is used for the printed object, I would like to update the remaning weight stored in the Spoolman database.

Disane87, 2024-06-16:
Thank you @bavala3010 for this feature request! At the moment I'm not quite sure whether I want to implement this, because I see the responsibility more with Klipper/Moonraker. However, I could imagine that you could implement this better via an API call to Spoolman, as I currently only want to retrieve the data read-only. 

Having the data changeable would be technically possible, but whether it would make sense to do this via HA is something I really need to think about in order to take possible side effects into account.

https://donkie.github.io/Spoolman/#tag/spool/operation/Update_spool_spool__spool_id__patch

But I gonna think about this!

sethiele, 2024-06-30:
Hi @Disane87 (and thanks for all the work).
I also would like to have a service to send a "[use spool](https://donkie.github.io/Spoolman/#tag/spool/operation/Use_spool_filament_spool__spool_id__use_put)" to spoolman.
In my example... I have a Bambu Lab A1, and monitor the state in HA. When a print is done, i will reduce the rest of the spool in spoolman.
I think for a first version It will be fine if we have a service (e.g. `spoolman_use_spool`) and i can give it the parameter spool `id` and `use_weight` or `use_length`.

Disane87, 2024-07-01:
@sethiele @bavala3010 I've implemented a service `spoolman.patch_spool` which you can use to patch values for a spool. Please refer to the readme for usage.

This should land in a dev build, so you need to test ist with that particular dev release in HA.

Note: 
You need to delete and add the integration again for the service to be registered. But it would be nice if you test that if its really necessary.

Disane87, 2024-07-01:
:tada: This issue has been resolved in version 0.5.0-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.5.0-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2024-07-01:
:tada: This issue has been resolved in version 0.5.0-dev.2 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.5.0-dev.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

bavala3010, 2024-07-01:
It's working. Thanks! 
Works without removing the initial integration. I just downloaded the 0.5.0-dev.2 version and restarted HA.
Attention points:
- the service updates the spoolman database directly, but the synchronisation between spoolman and HA is only every X minutes
- in the example parameters for the service, the example for the ID field is: spool_123. But the IDs in spoolman are just numbers. So, better put 123 as example.

Since my Bambulab X1C does not have Klipper, I will now try to make some automations that use the info from the BambuLab integration (like print weight and the spool serial number of the spool currently in use) to update the spoolman DB via this spoolman integration.

Disane87, 2024-07-02:
:tada: This issue has been resolved in version 0.5.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.5.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 881a76bf (2024-07-01): feat: service integration to change a spool in Spoolman via API

closed #119
commit 6b821b6f (2024-07-01): fix: removed periods from descriptions and names

closes #119
closed 2024-07-02 by Disane87