Disane87/spoolman-homeassistant #121: Starting with 2024.6 - Detected blocking call to open inside the event loop

Snuffy2, 2024-06-07:
### System Health details

## System Information

version | core-2024.6.1
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.31-v8+
arch | aarch64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4991
Installed Version | 1.34.0
Stage | running
Available Repositories | 1473
Downloaded Repositories | 73
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | February 3, 2025 at 7:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
remote_server | us-east-1-1.ui.nabu.casa
certificate_status | ready
instance_id | 94420c6a75f748f194d19c29af975a5f
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 12 (bookworm)
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 219.6 GB
disk_used | 75.8 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Mosquitto broker (6.4.1), Z-Wave JS UI (3.7.1), Victoria Metrics (1.9.1), Samba Backup (5.2.0), Grafana (10.0.0), eufy-security-ws (1.8.0-2), ESPHome (2024.5.5), Zigbee2MQTT (1.38.0-1), PS5 MQTT (1.3.3), go2rtc (1.9.3), Matter Server (6.1.0), Tesla HTTP Proxy (2.2.7), Govee to MQTT Bridge (2024.04.29-30cf7732)

</details>

<details><summary>keymaster</summary>

zwave_integration | zwave_js
-- | --
network_status | off

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 35
views | 23
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 28, 2024 at 3:56 PM
-- | --
current_recorder_run | June 7, 2024 at 6:20 PM
estimated_db_size | 2308.46 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

This is appearing in the logs multiple times.

### Reproduction steps

Upgrade to 2024.6

### Debug logs

```text
2024-06-07 18:18:54.692 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'spoolman' at custom_components/spoolman/sensor.py, line 226: image.save(image_path) (offender: /usr/local/lib/python3.12/site-packages/PIL/Image.py, line 2456: fp = builtins.open(filename, "w+b")), please create a bug report at https://github.com/Disane87/spoolman-homeassistant/issues
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 255, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 411, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 165, in async_update_listeners
    update_callback()
  File "/config/custom_components/spoolman/sensor.py", line 126, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1000, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1126, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1084, in __async_calculate_state
    if (entity_picture := self.entity_picture) is not None:
  File "/config/custom_components/spoolman/sensor.py", line 209, in entity_picture
    return self.generate_entity_picture(color_hex)
  File "/config/custom_components/spoolman/sensor.py", line 226, in generate_entity_picture
    image.save(image_path)
```
```

Disane87, 2024-06-09:
:tada: This issue has been resolved in version 0.4.4-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.4.4-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2024-07-02:
:tada: This issue has been resolved in version 0.5.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.5.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #122 (2024-06-09): Fix blocking of entity picture generation
Moves entity picture generation out of the event loop.

Fixes #121
commit 6f541777 (2024-06-09): fix: Fix blocking of entity picture generation

fixes #121
closed 2024-07-02 by Disane87