Disane87/spoolman-homeassistant #150: Support for multi colored filaments

Disane87, 2024-10-07:
### Discussed in https://github.com/Disane87/spoolman-homeassistant/discussions/148

<div type='discussions-op-text'>

<sup>Originally posted by **thewolfman56** October  1, 2024</sup>
Whenever I add a multi-colored filament to Spoolman, the new spool shows up with just the nozzle icon and the theme color.  Is there something I need to do so that colors show up similar to the single-colored filaments?  Multi-colored filaments look just fine in Spoolman itself.</div>

thewolfman56, 2024-10-18:
Just saw this in the Home Assistant Logs for my spool # 9 and #13 (which are multi-colored filaments) when I restarted Home Assistant after upgrading to 2024.10.2.

```
This error originated from a custom integration.

Logger: custom_components.spoolman.sensor
Source: custom_components/spoolman/sensor.py:70
integration: Spoolman (documentation, issues)
First occurred: 8:55:40 AM (2 occurrences)
Last logged: 8:55:40 AM

SpoolManCoordinator: Spool with ID '9' has no color_hex set. Can't create entity picture.
SpoolManCoordinator: Spool with ID '13' has no color_hex set. Can't create entity picture.
```

Disane87, 2024-10-26:
Thank you! Will investigate it. It seems color_hex is set on spools which had a color before the implementation of multi color in spoolman. New spools only with multicolor doesn't seems to have this. 

Will consider this when I implement multi color support

Disane87, 2024-12-16:
:tada: This issue has been resolved in version 0.7.0-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.7.0-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2024-12-16:
:tada: This issue has been resolved in version 0.7.0-dev.2 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.7.0-dev.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2024-12-16:
:tada: This issue has been resolved in version 0.7.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.7.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit bbd7025a (2024-12-16): feat: added multicolor images for multicolor filaments

closed #150
commit d2f28098 (2024-12-16): feat: added longitudinal color

closed #150
closed 2024-12-16 by Disane87