Disane87/spoolman-homeassistant #157: Spools are no longer available

BenGeba, 2024-11-12:
### System Health details

## System Information

version | core-2024.11.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.54-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4990
Installed Version | 2.0.1
Stage | running
Available Repositories | 1470
Downloaded Repositories | 28

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 6.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (6.6.0), File editor (5.8.0), Advanced SSH & Web Terminal (19.0.0), Get HACS (1.3.1), Mosquitto broker (6.4.1), Node-RED (18.1.1), MQTT Explorer (browser-1.0.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 19
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 2. November 2024 um 21:11
-- | --
current_recorder_run | 11. November 2024 um 23:35
estimated_db_size | 385.57 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Occasionally the error occurs that the HA integration cannot connect to the API. An internal server error occurs. After a new fetch (after 15 minutes) it usually works again. The error occurs rather sporadically. According to log details 14x within ~22 hours.

I will submit more DEBUG logs later. Since I activated it first, only the ERROR was visible in the log.

### Reproduction steps

1. Don't know how to reproduce the error

### Debug logs

```text
2024-11-12 21:41:01.210 ERROR (MainThread) [custom_components.spoolman.coordinator] Error fetching Spoolman data data: Error fetching data from API: 500, message='Internal Server Error', url='http://192.168.178.8:8000/api/v1/spool?allow_archived=False'
```

Disane87, 2024-11-12:
Thank you for that issue! It seems that spoolman is occasionally not available (because of that 500 error) or HA can't connect to it. That's not a problem of the integration nor I can mitigate this. The only thing in can do is to throw that exception.

You need track that error down with spoolman and check the connectivity. Please check the logs of spoolman if there are errors.

Sorry!

Disane87, 2024-11-13:
Thank you for your reply! I gonna close this. Bit you can keep us informed what the reason was (or reference this issue at the spoolman repo)

closed 2024-11-13 by Disane87