Disane87/spoolman-homeassistant #170: Add configurable entity id prefix to avoid entity id naming conflicts.

miguelangel-nubla, 2024-12-29:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Currently lots of integrations use the naming space as their own, so naming conflicts are common and complicate further automations.

### Describe the solution you'd like

Be able to configure a prefix like "spoolman" so "Spool1" creates sensor `sensor.spoolman_spool1` instead of the current `sensor.spool1`.
That way I can automate for `sensor.spoolman_*` instead of specific entity IDs.

### Describe alternatives you've considered

I understand there are ways to look for entities generated per integration such as using `integration_entities(integration)` in a template but a prefix is still easier and cleaner to keep entity IDs meaningful.

### Additional context

NA

Disane87, 2024-12-29:
Thank you @miguelangel-nubla 

Looks like a reasonable request. Will consider it if I'm ok the code next time ☺️

Disane87, 2025-03-08:
@miguelangel-nubla I dont think the entity prefix should be customized since the "namespace" should be only reserved by this integration. I guess this will land in the next public release.

Disane87, 2025-03-09:
:tada: This issue has been resolved in version 1.0.0-dev.2 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v1.0.0-dev.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2025-03-09:
:tada: This issue has been resolved in version 1.0.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v1.0.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 0b9fc063 (2025-03-09): fix: Added paragraph about naming of the entity ids in the README

fixes #170
closed 2025-03-09 by Disane87