Disane87/spoolman-homeassistant #174: Integration does not update HA entities on schedule

szymon3, 2025-01-08:
### System Health details

## System Information

version | core-2025.1.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/Warsaw
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4995
Installed Version | 2.0.2
Stage | running
Available Repositories | 1500
Downloaded Repositories | 35

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 43.6 GB
disk_used | 19.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.4.1), Zigbee2MQTT Proxy (0.2.0), Studio Code Server (5.18.0), Cloudflared (5.2.6), TheengsGateway (1.11.4), Folding@home (0.7.2), ESPHome Device Builder (2024.12.2), Terminal & SSH (9.16.0), Matter Server (7.0.0), Node-RED (19.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
[... 26 lines cut ...]

I realized that in Home Assistant entities are being updated only after HA Instance restarts. Scheduled update does not create newly added spool entities from Spoolman (created in the meantime).

### Reproduction steps

1. Do initial configuration of integration (spools will be imported into HA)
2. Add new spool in Spoolman and wait for scheduled interval
3. RESULT: New spools are not being created in HA

### Debug logs

```text
2025-01-08 12:39:41.440 DEBUG (MainThread) [custom_components.spoolman.coordinator] SpoolManCoordinator._async_update_data
2025-01-08 12:39:41.440 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: get_spools
2025-01-08 12:39:41.440 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: string_from_dictionary
2025-01-08 12:39:41.455 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: get_spools response [{'id': 1, 'registered': '2024-11-05T16:05:52Z', 'first_used': '2025-01-07T09:56:52Z', 'last_used': '2025-01-07T09:57:18Z', 'filament': {'id': 1, 'registered': '2024-11-05T16:03:50Z', 'name': 'PLA Black', 'vendor': {'id': 9, 'registered': '2025-01-08T10:19:48Z', 'name': 'Sunlu', 'external_id': 'Sunlu', 'extra': {}}, 'material': 'PLA', 'price': 40.0, 'density': 1.26, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 132.0, 'settings_extruder_temp': 230, 'settings_bed_temp': 55, 'color_hex': '000000', 'extra': {}}, 'price': 37.99, 'remaining_weight': 1000.0, 'initial_weight': 1000.0, 'spool_weight': 132.0, 'used_weight': 0.0, 'remaining_length': 329961.65720379987, 'used_length': 0.0, 'location': 'Dom', 'archived': False, 'extra': {}}, {'id': 2, 'registered': '2025-01-03T20:57:48Z', 'first_used': '2025-01-03T21:00:00Z', 'last_used': '2025-01-07T22:28:13Z', 'filament': {'id': 2, 'registered': '2025-01-03T20:55:27Z', 'name': 'PLA+ Black', 'vendor': {'id': 8, 'registered': '2025-01-07T16:12:14Z', 'name': 'JAYO', 'external_id': 'JAYO', 'extra': {}}, 'material': 'PLA+', 'price': 41.0, 'density': 1.24, 'diameter': 1.75, 'weight': 1100.0, 'article_number': 'XJJDE-PLA+-BK-1100G', 'settings_extruder_temp': 210, 'settings_bed_temp': 55, 'color_hex': '000000', 'extra': {}}, 'price': 41.0, 'remaining_weight': 607.1099999999999, 'initial_weight': 1100.0, 'used_weight': 492.89000000000004, 'remaining_length': 203554.0381841118, 'used_length': 165257.94317432903, 'location': 'Dom', 'lot_nr': '07041021E', 'archived': False, 'extra': {}}, {'id': 4, 'registered': '2025-01-07T16:09:32Z', 'filament': {'id': 4, 'registered': '2025-01-07T16:09:32Z', 'name': 'PLA+ Bone White', 'vendor': {'id': 7, 'registered': '2025-01-07T16:09:32Z', 'name': 'eSun', 'external_id': 'eSun', 'extra': {}}, 'material': 'PLA+', 'price': 55.0, 'density': 1.23, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 170.0, 'article_number': 'X00162006D', 'settings_extruder_temp': 220, 'settings_bed_temp': 60, 'color_hex': 'e0d7b6', 'external_id': 'esun_pla+_pla+bonewhite_1000_175_c', 'extra': {}}, 'price': 55.0, 'remaining_weight': 1000.0, 'initial_weight': 1000.0, 'spool_weight': 170.0, 'used_weight': 0.0, 'remaining_length': 338009.50250145356, 'used_length': 0.0, 'archived': False, 'extra': {}}, {'id': 5, 'registered': '2025-01-07T16:12:14Z', 'filament': {'id': 5, 'registered': '2025-01-07T16:12:14Z', 'name': 'Black', 'vendor': {'id': 8, 'registered': '2025-01-07T16:12:14Z', 'name': 'JAYO', 'external_id': 'JAYO', 'extra': {}}, 'material': 'PETG', 'density': 1.25, 'diameter': 1.75, 'weight': 1100.0, 'spool_weight': 120.0, 'settings_extruder_temp': 250, 'settings_bed_temp': 80, 'color_hex': '000000', 'external_id': 'jayo_petg_black_1100_175_n', 'extra': {}}, 'price': 40.0, 'remaining_weight': 1100.0, 'initial_weight': 1100.0, 'spool_weight': 120.0, 'used_weight': 0.0, 'remaining_length': 365861.4855075733, 'used_length': 0.0, 'archived': False, 'extra': {}}, {'id': 6, 'registered': '2025-01-07T16:12:51Z', 'filament': {'id': 6, 'registered': '2025-01-07T16:12:51Z', 'name': 'White', 'vendor': {'id': 8, 'registered': '2025-01-07T16:12:14Z', 'name': 'JAYO', 'external_id': 'JAYO', 'extra': {}}, 'material': 'PETG', 'density': 1.25, 'diameter': 1.75, 'weight': 1100.0, 'spool_weight': 120.0, 'settings_extruder_temp': 250, 'settings_bed_temp': 80, 'color_hex': 'FFFFFF', 'external_id': 'jayo_petg_white_1100_175_n', 'extra': {}}, 'price': 40.0, 'remaining_weight': 1100.0, 'initial_weight': 1100.0, 'spool_weight': 120.0, 'used_weight': 0.0, 'remaining_length': 365861.4855075733, 'used_length': 0.0, 'archived': False, 'extra': {}}, {'id': 7, 'registered': '2025-01-08T10:25:58Z', 'filament': {'id': 7, 'registered': '2025-01-08T10:20:59Z', 'name': 'Black', 'vendor': {'id': 8, 'registered': '2025-01-07T16:12:14Z', 'name': 'JAYO', 'external_id': 'JAYO', 'extra': {}}, 'material': 'Easy ABS', 'price': 44.0, 'density': 1.07, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 128.0, 'article_number': 'X001V9KIIV', 'comment': 'Pudełko X001V9KIIV\nASIN\t\u200eB0CJ33J51J', 'settings_extruder_temp': 235, 'settings_bed_temp': 70, 'color_hex': '000000', 'extra': {}}, 'remaining_weight': 552.0, 'initial_weight': 1100.0, 'spool_weight': 128.0, 'used_weight': 548.0, 'remaining_length': 214481.2446900812, 'used_length': 212927.03277203714, 'archived': False, 'extra': {}}, {'id': 8, 'registered': '2025-01-08T10:35:13Z', 'filament': {'id': 8, 'registered': '2025-01-08T10:33:54Z', 'name': 'White', 'vendor': {'id': 9, 'registered': '2025-01-08T10:19:48Z', 'name': 'Sunlu', 'external_id': 'Sunlu', 'extra': {}}, 'material': 'PLA Meta', 'price': 64.9, 'density': 1.24, 'diameter': 1.24, 'weight': 1000.0, 'spool_weight': 128.0, 'article_number': '5903899274000', 'color_hex': 'ffffff', 'extra': {}}, 'price': 65.0, 'remaining_weight': 91.0, 'initial_weight': 1000.0, 'spool_weight': 128.0, 'used_weight': 909.0, 'remaining_length': 60769.61087812793, 'used_length': 607028.3108595418, 'archived': False, 'extra': {}}, {'id': 9, 'registered': '2025-01-08T10:39:39Z', 'filament': {'id': 9, 'registered': '2025-01-08T10:39:39Z', 'name': 'Bambu Green', 'vendor': {'id': 10, 'registered': '2025-01-08T10:39:39Z', 'name': 'Bambu Lab', 'external_id': 'Bambu Lab', 'extra': {}}, 'material': 'PLA', 'density': 1.24, 'diameter': 1.75, 'weight': 1000.0, 'spool_weight': 250.0, 'settings_extruder_temp': 220, 'settings_bed_temp': 60, 'color_hex': '00AE42', 'external_id': 'bambulab_pla_bambugreen_1000_175_n', 'extra': {}}, 'remaining_weight': 157.0, 'initial_weight': 1000.0, 'spool_weight': 250.0, 'used_weight': 843.0, 'remaining_length': 52639.52824843201, 'used_length': 282644.0911683324, 'archived': False, 'extra': {}}, {'id': 10, 'registered': '2025-01-08T10:54:18Z', 'filament': {'id': 10, 'registered': '2025-01-08T10:47:09Z', 'name': 'Vertigo', 'vendor': {'id': 5, 'registered': '2024-11-05T16:01:57Z', 'name': 'Fiberlogy', 'extra': {}}, 'material': 'ABS', 'price': 88.4, 'density': 1.04, 'diameter': 1.75, 'weight': [... cut ...] :14Z', 'name': 'JAYO', 'external_id': 'JAYO', 'extra': {}}, 'material': 'PETG', 'density': 1.25, 'diameter': 1.75, 'weight': 1100.0, 'spool_weight': 120.0, 'settings_extruder_temp': 250, 'settings_bed_temp': 80, 'color_hex': '000000', 'external_id': 'jayo_petg_black_1100_175_n', 'extra': {}}, 'price': 40.0, 'remaining_weight': 1100.0, 'initial_weight': 1100.0, 'spool_weight': 120.0, 'used_weight': 0.0, 'remaining_length': 365861.4855075733, 'used_length': 0.0, 'archived': False, 'extra': {}}
```

Disane87, 2025-01-08:
Thank you filing this issue. This is a known bug atm and correlates with #103 

https://github.com/Disane87/spoolman-homeassistant/issues/103#issuecomment-2014885266

Need to refactor the entire system to get that done but unfortunately I don't have any time at the moment. It would be great if someone can assist.

Disane87, 2025-03-09:
Please check that again that should be fixed with #194. You can repopen this if it doesn't work for you

closed 2025-03-09 by Disane87