Disane87/spoolman-homeassistant #177: Spools not available when Klipper api error

Disane87, 2025-01-13:
I saw this issue too as I started using this last night. 

I edited coordinator.py to have the klipper check inside it's own try/except:

```
            try:
                klipper_url = config.get(KLIPPER_URL, "")
                if klipper_url is not None and klipper_url != "":
                    klipper_active_spool: int | None = await KlipperAPI(klipper_url).get_active_spool_id()
                    if klipper_active_spool is not None:
                        for spool in spools:
                            if spool["id"] == klipper_active_spool:
                                spool["klipper_active_spool"] = True
                            else:
                                spool["klipper_active_spool"] = False
            except Exception as exception:
                print("klipper api error")
```

_Originally posted by @floyduww in https://github.com/Disane87/spoolman-homeassistant/discussions/135#discussioncomment-11811988_

Disane87, 2025-01-13:
:tada: This issue has been resolved in version 0.7.0-dev.3 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.7.0-dev.3)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2025-03-05:
:tada: This issue has been resolved in version 0.8.0-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v0.8.0-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2025-03-09:
:tada: This issue has been resolved in version 1.0.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v1.0.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

closed 2025-03-09 by Disane87

rasasak, 2025-04-02:
After update to 1.0.0
I have a a problem again. IT stop showing filamenty, when klipper off ...

I try reinstall .. but still not showing

commit efeebeaa (2025-01-13): fix: improve error handling in SpoolManCoordinator for API calls

fixes #177