Disane87/spoolman-homeassistant #183: Request a spoolman update based on bambu HA integration

platini76, 2025-02-02:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

hello there is noone can adapt the scripts that are online to integrate this feature and sync spoolman with bambulab integration usage?

### Describe the solution you'd like

a simple integration to update automatically spools that are readed from ams, and a selector to select spool when no rfid codes

### Describe alternatives you've considered

I tried to use this script but something did not work for me:
https://www.reddit.com/r/BambuLab/comments/1hamev5/keeping_spoolman_uptodate_with_bambu_lab_and_home/

### Additional context

no additional

Disane87, 2025-02-02:
Hi! Thank you for the issue. But I don’t get where this is an issue of this integration. Could you please explain in detail what you are doing and what you expect? 

Additionally I don’t have a Bambu printer so I could help here unfortunately.

Disane87, 2025-02-03:
> hello DIsane and thanks for your kind reply,
> it can be a request of implementation... 
> I have a bambu and spoolman and I want to sync filament consumption... I think that for your integration is a little thing but for users it's a very good function

In don't want to implement some vendor specific stuff. Sorry! This would violate the responsibility of this integration. 

Yeah I have a Klipper integration, but this will be removed in later versions. 

There is a AMS sync and some other stuff already available. Sorry!

Disane87, 2025-02-03:
> vendor? this is no vendor..
> spoolman is a free and shareware spool database..

I know what spoolman is. But I don't want to implement specific APIs oder other stuff. My integration is only a bridge between spoolman and home assistant, nothing else.

Everything else (like synching Bambu spools to spoolman) shouldn't be done in my integration because this would be against the single responsibility of my integration, which is: show stuff from spoolman or trigger changes in spoolman. 

There are several solutions out in the wild to sync Bambu spools or uses lengths to spoolman. Or did I got you use case wrong?

Disane87, 2025-02-03:
Sorry in can't help you, because I don't have a Bambu printer and it's not really an issue of this integration. 

I guess a thread in the HomeAssistant forum or the Bambu integration itself would help you more.

platini76, 2025-02-03:
and I again thank you for your kind reply, 
all integrations say the same things... a tool where select the active spool, I see your readme... is there possibility to select the active spool on ams with some automation or script?

Disane87, 2025-02-03:
Sorry, but I don't habe any experience with Bambu and the entire ecosystem, so I can't answer your question about selecting a spool in AMS

closed 2025-03-03 by Disane87