Disane87/spoolman-homeassistant #186: Update extra attributes via patch action

tlanfer, 2025-02-16:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I have a bunch of very cheap zigbee humidity meters, cheap enough that i just throw one of them in each cereal box or vacuum bag. I would love to send their measured values as extra attributes to spoolman.
I configured an extra field in spoolman, and i get them as state attributes on my spools. But the patch action does not allow extra attributes.

### Describe the solution you'd like

I'd like to just be able to add extra attributes according to the spoolman API.
```yaml
action: spoolman.patch_spool
data:
  id: "{{ state_attr(entity_id, 'id') }}"
  extra:
    humidity: "{{ states('sensor.some_humidity_sensor') }}"
```

Something like this.

### Describe alternatives you've considered

I could probably update directly via http call instead.

### Additional context

Nothing

tlanfer, 2025-02-17:
Okay apparently i misjudged the issue. The action lets me set extra attributes, but the spoolman API is just weird with its types.
Looks like every extra attribute needes to be transmitted as string, but actaul strings need to be escaped _within_ that first string.
And i cant seem to convince HA to format it like that.

tlanfer, 2025-02-18:
Since the spoolman author says the API design is deliberate (https://github.com/Donkie/Spoolman/issues/630) i figured i should just wrap/unwrap the attribtues in the HA integration instead.
I'm not a python author, so i'm not 100% sure if this is the right way to do it. But it works on my machine :)

tlanfer, 2025-02-19:
In case enyone is curious. What this enables me to do is create two spoolman extra attributes: sensor (type text), humidity (type float). I then fill the attribute sensor with the name of a humidity sensor i have in Home Assistant.
Finally, i built an automation that runs on a schedule, that iterates over the spools in Home Assistant, checks if they have a sensor configured, and updates their humidity level if they do.

```yaml
repeat:
  for_each: "{{ integration_entities('spoolman') }}"
  sequence:
    - if:
        - condition: template
          value_template: "{{ state_attr(repeat.item, 'extra_sensor') is not none }}"
      then:
        - action: spoolman.patch_spool
          data:
            id: "{{ state_attr(repeat.item, 'id') }}"
            extra:
              sensor: "{{ state_attr(repeat.item, 'extra_sensor') }}"
              humidity: >-
                {{ states("sensor."+state_attr(repeat.item, 'extra_sensor')) }}
```

Having the humidity in spoolman lets me very easily visualize them not jsut in spoolman, but also in Home Assistant using a combination of auto-entities and multiple-entity-row:

[image]

Disane87, 2025-03-05:
Merged PR into dev. Thank you for this great addition! ✨

closed 2025-03-05 by Disane87