Disane87/spoolman-homeassistant #192: No entities in spoolman

b41571, 2025-03-01:
### System Health details

Core
2025.2.5
Supervisor
2025.02.4
Operating System
14.2
Frontend
20250221.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

i put in the right configuration (worked before).
the log viewer output shows that he finished his job successfully.

But there are no entities??

Integrationseinträge
spoolman
Keine Geräte oder Entitäten

in the dashboard i can see the the entities from a former installation but no value:

Entity not available: sensor.sunlu_red_pla.
 

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
2025-03-01 02:20:06.205 DEBUG (MainThread) [custom_components.spoolman.coordinator] SpoolManCoordinator._async_update_data
2025-03-01 02:20:06.205 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: get_spools
2025-03-01 02:20:06.205 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: string_from_dictionary
2025-03-01 02:20:06.214 DEBUG (MainThread) [custom_components.spoolman.classes.spoolman_api] SpoolmanAPI: get_spools response []
2025-03-01 02:20:06.215 DEBUG (MainThread) [custom_components.spoolman.coordinator] Finished fetching Spoolman data data in 0.011 seconds (success: True)
```

kellbot, 2025-03-02:
I updated HACS today and also lost all my Spoolman sensors. They all have the red icon denoting that the sensor is unavailable.

Disane87, 2025-03-03:
Have the same. Investigating it. Will report back 🙂

kellbot, 2025-03-03:
I haven't had a chance to dig into the logs, but I did verify that Spoolman is indeed providing a name and material in the API response. I am running HA in a docker container. I was running 2024.12 and updated HACS to 2.0.5 when I stopped getting Spoolman sensors. I have since updated to 2025.2, and tried rolling back to a few different versions of the Spoolman HA integration (as far back as 0.5) - no luck. Unfortunately I did all of this in the least scientific way possible.

kellbot, 2025-03-04:
I did a little bit of poking around today and confirmed that the data is coming in as expected from the Spoolman API. Everything looks good there and it creates a bunch of spools, but then they never get added as sensors. No errors that I can find from HA.

I'm wondering, is it possible I've got too many spools and it's timing out? I have over 130 logged, which is a whole bunch of sensors to create...

Disane87, 2025-03-05:
> I'm wondering, is it possible I've got too many spools and it's timing out? I have over 130 logged, which is a whole bunch of sensors to create...

It shouldn't be related to the number of spools. I personally only have around 30 spools, and it's also not working for me anymore. I'm currently investigating the issue, but it will take some time as I'm a bit under the weather and in the middle of renovating my house.

kellbot, 2025-03-05:
I found it! Or at least part of it. in sensor.py, line 55, if it can't add the sensor it is currently exiting the loop entire (return) instead of moving on to the next spool (continue).
I'm not set up to submit a PR at the moment, but it's an easy enough fix.
I'm not sure what it doesn't like about that spool, but I'm getting closer.

kellbot, 2025-03-05:
Aha, found it - the error was exactly as stated (a spool was missing the material) but the reported ID is the _filament_ ID, not the spool ID. Once I was looking in the right place I updated that filament with material information, and all my sensors are back.

In summary - sensors.py needs a `continue` instead of `return` in the loop that checks for names and materials, and the error message should say "Filament with ID" not "Spool with ID"

Disane87, 2025-03-06:
That seems reasonable. Will fix that. Thank you for your investigation ✨

Disane87, 2025-03-09:
:tada: This issue has been resolved in version 1.0.0-dev.1 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v1.0.0-dev.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Disane87, 2025-03-09:
:tada: This issue has been resolved in version 1.0.0 :tada:

The release is available on [GitHub release](https://github.com/Disane87/spoolman-homeassistant/releases/tag/v1.0.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit dc059365 (2025-03-09): fix: Iimproved logging for missing attributes

BREAKING CHANGE: Generation of spoolnames was fixed. It now has a preffix "spoolman_*" to avoid conflicts with other spoolnames and only tthe ID of the spool is used as a suffix. This means that the spoolname is now unique and can be used to identify the spool in the spoolman. This needs a reconfiguration of the integration and adjustsments in automation when spools are used.

fixes #192 #170
closed 2025-03-09 by Disane87