Disane87/spoolman-homeassistant #214: Error testing spoolman url: [Errno 104] Connection reset by peer

banter240, 2025-05-10:
### System Health details

<html>
<body>

Version | core-2025.5.1
-- | --
Installation type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual environment | false
Python version | 3.13.3
Operating system family | Linux
Operating system version | 6.12.23-haos
CPU architecture | x86_64
Timezone | Europe/Berlin
Configuration directory | /config

</body>
</html>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hello, currently trying to get HA working with spoolman, but currently whenever I try to configure the integration I get Errno104 using the following host address "http://192.168.178.20:7912" any idea why? I can reach it on my browser without any issue.

I run the Spoolman Docker on the same Device as the HAOS (Qnap).

Whenever i enter the ip of spoolman:
[image]

### Reproduction steps

-

### Debug logs

```text
-
```

Disane87, 2025-05-20:
Is your HA available via HTTPS?

banter240, 2025-05-20:
Uhm, it is not, my entire HA / Docker Station / Virtualization Station (QNAP) are on http

Disane87, 2025-05-20:
Could you reach your spoolman instance via your url with `/api/v1/` at the end? Like: `http[s]://spoolmanurl/api/v1/info`

This should give you some information about the spoolman system

banter240, 2025-05-20:
Yes i can, the output is:

{
"version":"0.22.1",
"debug_mode":false,
"automatic_backups":true,
"data_dir":"/home/app/.local/share/spoolman",
"logs_dir":"/home/app/.local/share/spoolman",
"backups_dir":"/home/app/.local/share/spoolman/backups",
"db_type":"sqlite",
"git_commit":"1888240",
"build_date":"2025-03-18T20:55:07Z"
}

Disane87, 2025-05-20:
Is it possible that you spin up a simple Linux container interactively and do a cURL to that api endpoint? 

I guess the docker instances are isolated.