Disane87/spoolman-homeassistant #219: Subtract filament usage from spool during active print (Home Assistant + Spoolman)

jrspowers, 2025-06-16:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hello, I’m working with Spoolman in Home Assistant — I’ve got everything working, but there’s no way to subtract the filament used from the total amount. I tried using the filament_used sensor from Moonraker, but it’s not ideal since it resets to 0 when the print completes or hits 100% status, so the value isn’t stored anywhere. Does anyone have an idea?

### Describe the solution you'd like

Is there a way to still have this calculated and processed into Spoolman so that I can always see the values update live?

### Describe alternatives you've considered

I tried triggering it with the sensor at 96%, but that didn’t work properly either

### Additional context

Not applicable

jrspowers, 2025-06-16:
I can't manage to get the plugin working in Klipper via the moonraker.conf — it doesn't pass anything through, which is really strange. That's why I wanted to use the filament_used from Moonraker via Home Assistant, but unfortunately that's not possible. So that's why I'm asking this.

Disane87, 2025-06-24:
You could create an automation to store the value before it gets reset to 0, but that’s a bit of a workaround. The intended workflow is to let Moonraker handle the used filament consumption and report it to Spoolman — which generally works quite reliably.

jrspowers, 2025-06-24:
Yes, I completely understand that, but Moonraker doesn’t want to connect to my Spoolman — I don’t know why. But I still want to try using the automation. Could you send me how you think it should be stored?

closed 2025-07-10 by Disane87