Disane87/spoolman-homeassistant #224: Use active spool for bambulab AMS

platini76, 2025-07-04:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

could you think to use instead of klipper active spool some attribute from bambulab integration to sort the active or spool present in ams or ams pro?

### Describe the solution you'd like

modify the card where is klipper active spool

### Describe alternatives you've considered

no ohers

### Additional context

no additional

Disane87, 2025-07-04:
Thank you for your feature request! Unfortunately, I won’t be adding this feature. Let me explain why.

The current implementation of the active spool property in my integration is just a temporary workaround. This is because the Moonraker Home Assistant integration currently doesn't provide a proper "active spool" property.

You can see this limitation discussed here:
https://github.com/marcolivierarsenault/moonraker-home-assistant/issues/342

If that property is eventually added on the Moonraker side, my workaround will become obsolete (deprecated).

Why I don’t want to add this BambuLab-based feature here:
This integration is strictly intended to be a bridge between Home Assistant and Klipper. Nothing more. Adding support for BambuLab's AMS data or mixing in other printer ecosystems doesn't fit that scope.

If someone wants more advanced logic (like selecting or matching spools from other brands or systems), that should be handled in Home Assistant itself—using templates or automations. That’s the right place for this kind of customization.

platini76, 2025-07-04:
it wouldn't be vendor specific... because it would be coil specific, meaning IF it is recognized, it is highlighted, just like klipper does, just use a sensor like the serial... why can't you suggest a working code? can anyone do that?

Disane87, 2025-07-04:
Please test my suggestions with Bambu integration and the Bambu Sync to spoolman. With both you will get the desired result, even without any line of code. You only have to do some templating. 

I will close this issue, since it's not an issue with my integration. Feel free to open up a new issue for bugs or feature requests. 

Thank you!

closed 2025-07-04 by Disane87