Disane87/spoolman-homeassistant #226: Spoolman update time

jrspowers, 2025-07-09:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

faster update

### Describe the solution you'd like

Hi, I would like to ask if it’s possible to make Spoolman update every 15 seconds, because currently it takes 1 minute for changes to come through. This can be a bit long for certain automated processes. Thanks in advance

### Describe alternatives you've considered

faster update 

### Additional context

Hi, I would like to ask if it’s possible to make Spoolman update every 15 seconds, because currently it takes 1 minute for changes to come through. This can be a bit long for certain automated processes. Thanks in advance

Disane87, 2025-07-10:
I imagine reducing the update interval could put quite a bit of load on HA/Spoolman, especially if you have many spools configured. For that reason, I generally wouldn’t recommend going below one minute.

However, I should mention I don’t have much direct experience with that specific scenario. Could you share more about your use case? I’d love to understand what you're working on and why you need a faster interval. That way, I can better see what you’re trying to achieve and maybe help find a good solution.

jrspowers, 2025-07-10:
Hello, I’m using a display with openHASP showing all the spools with their weights. The active spool changes color from blue to green, but this currently takes 1 minute. That’s why I would like to make it shorter. Do you understand, or is there another way to do this?